InternRobotics/NavDP #21: Quick question about deploying on real robots + real-time inference

Disembark1, 2025-08-06:
Hey there！First off, thanks a ton for open-sourcing this project—I’ve been really into it lately.  

Quick question for you:  
1. Could this project actually run on a real robot? Like, not just in simulation or on a computer?  
2. If it can, does it handle real-time inference smoothly? You know, like processing inputs and making decisions fast enough for the robot to act in real time?  

No worries if it’s too much trouble—just curious how feasible this would be. Thanks again for your work on this！ 😊

wzcai99, 2025-08-06:
Hi, our NavDP can achieve good sim-to-real generalization without any fine-tuning. You can try with your own robots. As for the real-robot deployment,  we run our NavDP as a server in a local network and let the robot keep sending HTTP requests to query for the navigation trajectory. With a GeForce 4090 machine as the server, the inference speed can achieve around 10Hz.

wzcai99, 2025-08-07:
Yeah, this is almost the entire pipeline. 
As the NavDP generates trajectories, you still need another function mapping trajectories to robot control commands. This often can be done on the robot platform.

closed 2025-08-07 by wzcai99