InternRobotics/NavDP #25: traning && other robots

leo7611, 2025-08-11:
Thank you for your excellent work!

I want to know if you have open-source the model training code and the code for other robots (such as GO2) to run in simulation environments. I couldn't find it.

leo7611, 2025-08-11:
> Hi, the training code is open-sourced at InternNav, please check here https://github.com/InternRobotics/InternNav/blob/main/internnav/trainer/navdp_trainer.py. The evaluation of other robots is not open-sourced yet, please stay tuned.

I'm a bit confused. What is the relationship between InternVLA-N1 and NavDP? Can't NavDP run and deploy independently

wzcai99, 2025-08-11:
InternVLA-N1 is a dual-system which the system-1 is based on NavDP. NavDP can be trained independently, and we will update both the training code in this repo in the near future.