InternRobotics/NavDP #34: Launching navdp_server.py with a provided checkpoint fails during /navigator_reset because NavDP_Policy’s cond_pos_embed.position_embedding.weight in the checkpoint has shape [130, 384], while the current code constructs the module with [132, 384].

guactaco12, 2025-09-01:
Launching navdp_server.py with a provided checkpoint fails during /navigator_reset because NavDP_Policy’s cond_pos_embed.position_embedding.weight in the checkpoint has shape [130, 384], while the current code constructs the module with [132, 384].

I used the downloaded checkpoint after filling the google form, could you please let me know why this checkpoint is different from the NavDP_Policy model definition for cond_pos_embed.position_embedding?

guactaco12, 2025-09-01:
> Hi, did you download the newer checkpoint or not? There are two checkpoints provided after filling the Google form.

Thanks for the quick reply, do you mean using the `navdp-cross-model.ckpt` instead of `navdp-weights.ckpt` ?

wzcai99, 2025-09-01:
> > Hi, did you download the newer checkpoint or not? There are two checkpoints provided after filling the Google form.
> 
> Thanks for the quick reply, do you mean using the `navdp-cross-modal.ckpt` instead of `navdp-weights.ckpt` ?

Yes.

guactaco12, 2025-09-01:
Thanks a lot! That does solve the issue.

closed 2025-09-01 by guactaco12