InternRobotics/NavDP #35: Question on NoGoal Task

guactaco12, 2025-09-02:
Thanks for the great open-source work again. I experienced some issue in the indoor settings such as `internscenes commercial` with nogoal task. The robot always go to the corner of the room and will go through it and leads to a complete black image for input. Do I have something wrong with the physics for simulation?

guactaco12, 2025-09-03:
> Thanks for your valuable feedback. Can you provide some example videos?

Sure, here is a sample video. Please note that I have migrated Isaac lab to 2.2.0 and Isaac sim to 5.0.0.  I got it run after fixing some errors and warnings. The `cluttered_easy` scene seems ok but not the `internscenes`.

https://github.com/user-attachments/assets/c1e118f0-82af-4423-9f79-bfb7a8d17b21

wzcai99, 2025-09-03:
We will check this recently and reply to you later.

wzcai99, 2025-09-16:
We recommend to downgrade the isaacsim and use the isaacsim-4.2.

closed 2025-09-18 by wzcai99