InternRobotics/NavDP #39: imagegoal navigation

deavn2236, 2025-09-16:
Hello, navdp is a great job, but when I use your provided checkpoint for image-goal navigation testing, the success rate is almost 0. Is this normal? Does your provided checkpoint support image-goal navigation? If not, could you please provide a checkpoint that supports image-goal navigation?

deavn2236, 2025-09-17:
> Hi, which evaluation scenes are you using? If everything goes as expected, the imagegoal performance should be like this:
> 
> <img alt="Image" width="517" height="247" src="https://private-user-images.githubusercontent.com/115710611/489956236-3dbae6a8-5a7e-474f-8945-6540d5f683e4.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTgwNzYxMzIsIm5iZiI6MTc1ODA3NTgzMiwicGF0aCI6Ii8xMTU3MTA2MTEvNDg5OTU2MjM2LTNkYmFlNmE4LTVhN2UtNDc0Zi04OTQ1LTY1NDBkNWY2ODNlNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkxN1QwMjIzNTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01Y2RlNjI1YTczYzhjYTg0ZDMzZWZlMzZiMGE2NmNlMGVhYTlmZWVlNjM4NWEyMzE5MzA5ODU5NzE3MTlhOTg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ekU2hBYI8g7_50WqYQJ3pkjr9iKfGKjEkYVczJoSY38">
> And please check the technique report for more details. [report](https://internrobotics.github.io/internvla-n1.github.io/static/pdfs/InternVLA_N1.pdf)

Thank you for your prompt reply. I used the provided navdp_checkpoint.ckpt and ran the evaluation for NavDP using the following command:
python eval_imagegoal_wheeled.py --port 1888 --scene_dir ~/NavDP/assets/scenes/cluttered_easy --cluttered_easy 0 --scene_scale 1
However, the overall success rate I obtained for the easy0 scenario is only 0.1386 (presumably 13.86%). I am using the main branch of the code. Also, I noticed that in your paper, the reported results are based on the cluttered_easy environments, but currently I can only evaluate one specific scene (cluttered_easy n) per run.

Could you please advise how I should adjust these two aspects — (1) improving the low success rate, and (2) evaluating across the full cluttered_easy benchmark instead of just one scene?

Thank you!

deavn2236, 2025-09-19:
> Make sure you are downloading the latest checkpoint with the file name navdp-cross-modal.ckpt. And can you share one of the fps.mp4 results?

I have tried other scenes such as easy2, easy3, and internscenes, but the highest success rate I achieved is only 30+%. Is this due to an issue with my environment configuration, or is it related to parameter tuning?

deavn2236, 2025-09-25:
> Hi, I found a mismatch of one parameter between our evaluation code and the released code. Now I have pushed the latest code, please try it.

Thank you for your reply. After modifying the parameters, the evaluation performance of NavDP has indeed improved, achieving around a 50% success rate in the "easy" scenarios. However, I still have a few questions:

The paper mentions that the model possesses camera and platform adaptability. However, why does adjusting the camera position lead to such a significant change in performance? Were datasets like "cluttered_easy" included in the training data?
I also tried evaluating other methods such as NoMaD and GNM. GNM encountered model loading issues, and even after adjustments, it stopped after only a few steps. NoMaD’s success rate also remained below 10%. In their respective papers, methods like NoMaD construct an image-based topological graph from start to goal, which significantly boosts ImageGoal navigation success rates.
A single NavDP evaluation currently takes about one hour. The code for setting multiple environments via "num_envs" seems incomplete—how can I efficiently evaluate across multiple environments simultaneously?
I would also like to use the NavDP platform to evaluate my own method. I would greatly appreciate your response. Thank you very much!

wzcai99, 2025-09-26:
Thank you for your questions. 
1. All evaluation scenes are not included in generating the training data. And you can see the NavDP can successfully deal with the out-of-distribution scenes. As for adjusting the camera position lead to performance change, this is because this parameter influence the capture image-goal, and this can lead to a discrepancy between the observation images and the image-goal.  Current version of training only support sample image-goal and the observation images within the same training trajectory.
2. I agree with that. But considering the scenarios that robot navigation in unknown scenes, it is not always possible to build a graph in advance and then navigate. 
3. We will complete the support for multi-env evaluation in the near future.

wzcai99, 2025-09-26:
The navdp-cross-modal.ckpt was trained on the full version of InternData-N1 (subset vln-n1). And to collect the start poses and goal positions, we convert the scene usd into esdf map and perform oracle path-planning between random sampled pairs of points. Then we keep the traversable pair samples and record the initial orientation based on the path planning result.

deavn2236, 2025-09-27:
> The navdp-cross-modal.ckpt was trained on the full version of InternData-N1 (subset vln-n1). And to collect the start poses and goal positions, we convert the scene usd into esdf map and perform [...]

Thank you! Could you please provide the code for collecting the start poses and goal positions? I would greatly appreciate it.

wzcai99, 2025-10-17:
@deavn2236 Hi, sorry for my late reply. Here is a basic implementation about how to collect the start poses and goal positions. 

[process_spawn.py](https://github.com/user-attachments/files/22961876/process_spawn.py)

zlm646, 2025-10-20:
Hi, how do you launch the evaluation process? I run python eval_pointgoal_wheeled.py --port 8888 --scene_dir ./assets/scenes/cluttered_easy/ --scene_index 0 --scene_scale 1.0 and I got:
File "~/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/from_files/from_files.py", line 222, in _spawn_from_usd_file
    raise FileNotFoundError(f"USD file not found at path: '{usd_path}'.")
FileNotFoundError: USD file not found at path: ''.